ECESeniorDesign/greenhouse_envmgmt #5: Fertilizer and Water Quality Levels

jeffrobots, 2015-12-19:
After doing some research into various methods for determining soil characteristics such as fertilizer and pH levels, it seems like it's going to be a lot easier to measure pH OR conductivity of our water reservoir, as this can be done using a probe in the water bucket whereas measuring soil quality is generally done by taking soil samples and diluting them with a known solution - it's not exactly practical at this point with our budget.

Note that I say OR above because getting a reliable sensor can be pretty expensive for either method.

#### pH
Measuring pH can be done with a glass probe and an interface circuit containing a gain stage, filtering, and potentially temperature compensation. BNC connected probes of varying quality can be had on Amazon for less than 30 dollars. This one seems like a no brainer: I even found a resource [here](http://www.eecs.ucf.edu/seniordesign/sp2014su2014/g09/documents/Group-9-Senior-Design-1-LeafAlone-Hydroponics.pdf) that may have some useful information on how to interface to one of these things. It looks like this project got a lot of it's info from Atlas Scientific, so I'll check that out to see what they have to offer.

#### Conductivity
A similar thing with conductivity can be done using a toroidal non-contact conductivity sensor. The 
5000TC Toroidal Conductivity Sensor is an example of this. I'm having a hard time finding prices, but they look to be somewhere around $80-$100 dollars a piece. The downside here is that while conductivity might give us a guess as to nutrient levels (assuming we WANT to use water based fertilizers), it really just measures everything, so it's going to end up being dependent on whatever the water supply is. High chlorine levels would essentially look the same as high fertilizer levels, for example.

cconklin, 2015-12-19:
I wonder how the Flower Power was able to do it so easily?

jeffrobots, 2015-12-19:
In short... I don't think it was. I think what they did is use a cheap electrode to measure conductivity of the soil and then they just called it good. I have a feeling that the sensor on the Flower Power more or less amounted to a really crude, rough guess as to the soil condition.

Even having air pockets in the soil would throw off a simple electrode like that - the measurement would be highly dependent on water content.

cconklin, 2015-12-19:
I guess my feeling is a crude guess about the soil condition is better than an accurate understanding of the water condition (since the plants won't be in the reservoir water). Ignoring air pockets, would it be possible to combine soil moisture settings and conductivity to control for the effect of water? I would be happy to help with the constructing of a model for that once we get back from break.

Also, since pH is determined mostly by a chemical rather than electrical process, I say we go with conductivity.

jeffrobots, 2015-12-19:
I suppose it's possible to control for moisture effects. I've not been able to find any off the shelf soil conductivity sensors though. The DIY type stuff seems kind of ghetto and not very elegant, but perhaps we could come up with something.

What do you mean by chemical rather than electrical process? You can measure pH with an electrode probe. Here are a few references. [How-To guide using probes and a simple interface](http://www.allaboutcircuits.com/textbook/direct-current/chpt-9/ph-measurement/) | [Example of such a probe on Amazon](http://www.amazon.com/dp/B001EHJO64?psc=1)

Conversely, Here's an example of a conductivity meter [from Amazon](http://www.amazon.com/Sper-Scientific-Conductivity-Benchtop-Salinity/dp/B00BEZX6Z6/ref=sr_1_2?s=industrial&ie=UTF8&qid=1450552958&sr=1-2&keywords=conductivity+probe)

If we want to create an in-soil conductivity meter, we're probably going to have to make it ourselves.

cconklin, 2015-12-19:
I stand corrected.

cconklin, 2015-12-19:
Could we get something like [this](http://www.amazon.com/Luster-Leaf-1880-Rapitest-Electronic/dp/B000HHLJQA) and use an ADC to get our readings (since we aren't using many of our GPIO pins)?

jeffrobots, 2015-12-19:
That's certainly an interesting idea. My main concern there would be reverse engineering it and getting useful data from it. It's a slightly less elegant approach, but it could work. I find it really odd that just the probes themselves cost more than these all-in-one units. Perhaps I'm focusing too much on making the solution elegant and one-of-a-kind.

For the soil moisture level, [we can use these.](https://www.sparkfun.com/products/13322) 

The ADC isn't an issue. I have three 4ch I2C ADCs mounted on some surfboards already. We've got plenty of power to go around there even without touching more GPIO.

jeffrobots, 2015-12-19:
According to the reviews on the types of meters, meters like this apparently corrode over time and become inaccurate. Something to think about I suppose. There probably isn't a way around that other than finding a way to clean them or just replacing them. If you think it's a good route forward, I can order a few to be here after break and we can hack a few of them apart. My plan was to try and figure out the soil moisture stuff over break along with how to solve the issue I've created with the humidity sensors in issue #4. Oops...

cconklin, 2015-12-20:
I think it is an interesting way to go forward. Let's do some research into what's out there for those, and try to have a list of candidates in the next week. If we like any of the options, we can order a couple and hack them apart.

jeffrobots, 2015-12-20:
If we want to do just pH, [we could use this one](http://amzn.com/B0000DI848) which is probably the cheapest bet. It's $25 for a 6 pack. It's a great blend of super cheap and replaceable and *allegedly* accurate readings. 

For fertility, we could use the one you linked or [this one](http://amzn.com/B006JFY9ZW)

cconklin, 2015-12-20:
Let's get the pH meter you linked. While the one I found could measure more things, I don't think it could do it all at once, and developing the circuitry to switch between modes would be tricky.

jeffrobots, 2016-01-04:
I decided to hold off on ordering the stuff until I get back to school since I haven't been able to work with the interface modules that much. Fortunately, I just got a new soldering station with a hot air gun and SMT tips, so hopefully I can repair my earlier screwup and get everything else assembled.

jeffrobots, 2016-01-19:
After reading some literature and reviews, it seems that the lifetime of an anode/cathode type sensor like a pH sensor or soil moisture sensor can be extended significantly by keeping them powered off when you aren't using them. This would be easy to accomplish with a single MOSFET.

commit 40a2bbd3 (2016-02-03): Close #5: Added soil moisture sensing

Soil moisture sensors now have full support. I'm closing this issue as
we may be removing pH sensing from the project for now, though space
will exist on the ADC to measure an additional parameter.

As of this update, the plant sensor prototype is functionally complete.
Calibration may be needed to dial in the soil moisture sensor when it's
used in soil, but this will be a simple mapping function if it is even
required. The sensor currently outputs 0 for dry air and 1 when
submersed in water.
closed 2016-02-03 by jeffrobots