ECESeniorDesign/greenhouse_envmgmt #6: Integrate sensors with webservice

jeffrobots, 2016-01-15:
[See issue 1 on Webservice Demo for more info](https://github.com/ECESeniorDesign/greenhouse-webservice-demo/issues/1)

Need to create an interface for linking webservice to the available sensor hardware.

My current idea is to create a model of a sensor cluster that contains information on current values and addresses for the various sensors.

cconklin, 2016-01-15:
As I see it, you intend to have a class that knows all about getting sensor information. That should work: all you would need to do would be to add a method of knowing which plant the sensors are for, and a `saveAllSensors` method that does the following:

```python
def saveAllSensors(self):
    # Assuming that you imported models
    plant = models.Plant.for_slot(self.plantId)
    # Save light data
    plant.sensor_data_points.light().build(sensor_value=self.lux).save()
    # More saving here...
```

jeffrobots, 2016-01-15:
The latest commit hopefully is doing what you're talking about. Aside from the greenhouse demo not having a space for a temperature record, this is hopefully ready to go into an implementation that can be used to demo. 

I still need to tie in the I2C multiplexer and get the ADC/acidity/soil moisture stuff figured out. I also need to fix the humidity sensor situation. Aside from these glaring, very important issues, having this interface somewhat fleshed out is going to make tying these other modules in a bit more clearcut.

jeffrobots, 2016-01-15:
I forgot to add that the method for determining which sensor belongs to which plant is pretty much going to be to hardcode it in at the time of SensorCluster object instantiation. I could add a dictionary that would make it easier to read, but at the end of the day the addresses will still need to be hardcoded.

cconklin, 2016-01-15:
I just added temperature. You can look at the diff at ECESeniorDesign/greenhouse-webservice-demo@a101052 to see how to add more (if you need to).

cconklin, 2016-01-15:
Make sure that you also open a database connection and that you `import models` from the greenhouse webservice when you do it for real.

jeffrobots, 2016-01-23:
As per our test yesterday, I'm going to close this issue just to clear up the panel. However, since my code isn't currently interfacing outside of the tests to verify that it will work, you should be sure to get documentation up so that I can wrap all of that into the relevant scripts when the time comes to do so.

closed 2016-01-23 by jeffrobots