ECESeniorDesign/greenhouse_envmgmt #8: Design PCB for sensor modules

jeffrobots, 2016-01-21:
KiCad will be used to develop sensor module PCB files as soon as the prototype circuits are proven. The packages will use the smallest sensor packages possible. They should also include necessary pull-up resistors and bypass capacitors. The datasheets contain layout information (as well as connection information) and should be referenced often. 

- [ ] Proof the sensor module prototype in its entirety. 
- [ ] Move sensor prototype to a soldered perfboard and proof again
- [ ] Move design to schematic and PCB Gerber file
- [ ] Design review of PCB layout
- [ ] Order and test received PCBs

jeffrobots, 2016-01-24:
@cconklin In case you're interested, [image]. 

Even if you're not interested, I think I should go over the schematics with someone in the department before I send them off for fabrication. That's a little ways away though. I still need to get some new components in just to verify everything. I also need to figure out if any sort of interface/amplification will be needed on our pH and soil moisture meters.

cconklin, 2016-01-24:
Agreed. We can discuss it further with Derefinko on Monday.

jeffrobots, 2016-02-08:
The last required tasks before the PCB can be ordered are verifying the BJT soil sensor power control as well as looking into adding status LEDs where appropriate.

jeffrobots, 2016-02-10:
As of 6536d1a24c97237ee98df545fb8141bdaf6287d7 I have confirmed that using a transistor to power the soil sensor on and off will work. 

I will be reworking the PCB files tomorrow to include this and a status LED.

jeffrobots, 2016-02-13:
@cconklin I just uploaded the gerbers to OSHPark. I'll figure out how to order through the department on monday. 

If you want to see it, take a look here. 

[image]

jeffrobots, 2016-03-03:
@cconklin 

Everything works perfectly on the board except for, as you might be able to guess, the light sensor. 

The status LED isn't working, and I thought it was an issue of the resistor size. After closer inspection, it seems possible that I actually wired it up backwards. They don't really have very visible markings for determining which way to put it and I wasn't really thinking straight when I was soldering it. 

Are you free at all tomorrow afternoon in case I need a second pair of eyes on the light sensor? I've looked it over a bit and honestly have no idea what its problem is.

cconklin, 2016-03-03:
I probably won't be available.

jeffrobots, 2016-03-03:
After testing continuity on the pads of the light sensor, I discovered that the SMBus pads on the light sensor are indeed swapped. Luckily, @cconklin convinced me to put jumper pads so that I can switch them around if need be. I grabbed a few jumpers from the lab and will stick them on the board and cut the traces as soon as possible.

jeffrobots, 2016-03-03:
Closing as per today's testing. 

Remaining issues regarding the sensor PCB will be design enhancements and should have their own issues.

closed 2016-03-03 by jeffrobots