ECESeniorDesign/greenhouse_envmgmt #18: Can we instantiate multiple SMBus objects?

cconklin, 2016-02-24:
I'm working on the controls service on the web server, and I realized that I will need to have the controls establish a connection to the SMBus, which in production will occur at the same time as the sensor program (which also has a connection to the bus) is running. Will this cause a problem?

jeffrobots, 2016-02-25:
I ran a few tests to see how the SMBus class will handle multiple bus objects. It doesn't appear that you're going to run into any issues using multiple bus objects. I haven't tried anything with mutex/locks which I know you will be using, so I can't really make any guarantees there, but I suspect there's no issue either way.

Just make sure you always instantiate it as such: 

```python
SensorCluster.bus = smbus.SMBus(1)
ControlCluster.bus = smbus.SMBus(1)
```

cconklin, 2016-02-25:
Thanks for looking into it.

closed 2016-02-25 by cconklin

jeffrobots, 2016-02-25:
@cconklin 

I wanted to note that the SMBus objects have a close() and open() method. I've not used them a whole lot, but it appears that close() acts like a sort of pseudo-destroyer for a bus object. I say pseudo because while it does indeed disconnect the object from the I2C bus, it doesn't look like the open() method actually works right. Once you close it, reconnecting the object doesn't make it work again.

If this is functionality that you need, you'll need to play around with it to see if it will actually work properly.