ECESeniorDesign/greenhouse_envmgmt #19: Making code available to web service

cconklin, 2016-02-26:
We are going to need a portable and simple way to ensure that the web service has access to the sensor and control code in this repository. I'm thinking that we could package this up and put it in PyPi so it is a simple "pip install" to get the necessary behavior anywhere.

Information about putting code on PyPi can be found [here](http://peterdowns.com/posts/first-time-with-pypi.html).

cconklin, 2016-02-26:
I can make the necessary changes, though it will result in a slight restructuring of the repository.

jeffrobots, 2016-02-26:
If you want to make the changes, I'm fine with that. I'm sure it would take me a day or two to get it figured out and ready for PyPi, since I'm not familiar with it. I'm assuming you will need to hide or remove the hardware aspect of the repo?

Either way, I'll look over the information you linked so that I at least know what's going on and can contribute.

jeffrobots, 2016-02-27:
I think I follow what needs to happen. 

Basically, sensor_models and control_models need to put placed in their own directories, and then a new directory called "utils" would also need to be created to handle the utility functions that I've added. Each directory gets it's own `__init__.py`, the repo gets a new README, and then I tag for release. It doesn't sound too hard if you prefer I try and take care of it.

jeffrobots, 2016-02-27:
@cconklin 

Greenhouse environment management (`greenhouse_envmgmt`) has been added to PyPi at release tag v1.0 with 146f08a082325edbb9c3235b110431809f9555db

After doing the pip install to test it out, I realized I'm going to need to clean up import dependencies. Think you could help out with that?

cconklin, 2016-02-27:
I can help with that.

cconklin, 2016-02-27:
It looks like your setup.py is bugged. There is a syntax error, and my pip can't find the python-smbus module.

```
Complete output from command python setup.py egg_info:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/var/folders/jh/vszcshj577lcb234jny1fjkr0000gn/T/pip-wCAMGt-build/setup.py", line 10
     keywords = ['i2c', 'sensors', 'controls'],
            ^
SyntaxError: invalid syntax
```

Fixing the syntax error, I get:

```
Collecting python-smbus (from greenhouse-envmgmt==1.0)
  Could not find a version that satisfies the requirement python-smbus (from greenhouse-envmgmt==1.0) (from versions: )
No matching distribution found for python-smbus (from greenhouse-envmgmt==1.0)
```

jeffrobots, 2016-02-27:
I forgot to update setup.py with the one I used to register and upload with PyPi. 

I removed install-requires completely in this version, but if you'd like to include it, the package is just smbus. 

Sorry for the confusion.

cconklin, 2016-02-27:
Are you still having trouble with import dependencies?

jeffrobots, 2016-02-27:
The issue is more than - in order to use anything from the greenhouse-webservice, it needs to be added to the pack. It's more irritating than code-breaking. There are also probably packages being imported multiple times. It's something that can be cleaned up at a later date I believe.

jeffrobots, 2016-04-12:
Closing. The PyPi package will need to be updated once the sensors are calibrated, but is currently available.

closed 2016-04-12 by jeffrobots