ECESeniorDesign/greenhouse_envmgmt #23: Production controls circuitry and enclusure

jeffrobots, 2016-03-22:
The controls perfboards need to be finished and added to the control box. 

Putting together the rest of the controls circuitry is dependent on the enclosure and cable management that will be used. 

Tasks:
- [ ] Determine cable routing and plug placement
- [ ] Arrange and mount remaining controls hardware
- [ ] Mount hardware and cabling inside enclosure with the raspberry pi

cconklin, 2016-03-22:
See this [micro-usb connector for the Pi](http://www.amazon.com/gp/product/B00OK6G036/ref=pd_lpo_sbs_dp_ss_1?pf_rd_p=1944687762&pf_rd_s=lpo-top-stripe-1&pf_rd_t=201&pf_rd_i=B00UJ967GS&pf_rd_m=ATVPDKIKX0DER&pf_rd_r=0JQS6Q9ACYVMTJW2FPH3). We will also need a 12V to 5V [buck converter](http://www.amazon.com/Nextrox®-Display-Supply-Converter-Module/dp/B00CBCGAL8/ref=pd_sim_23_2?ie=UTF8&dpID=410nEQxN-pL&dpSrc=sims&preST=_AC_UL160_SR160%2C160_&refRID=197WC55HT3SFKKTYTAV1).

jeffrobots, 2016-03-23:
As per recent work:

The controls circuitry has been mounted on a perfboard as the final solution. I have yet to verify that everything works after a few recent changes on the control board. The relay board appears to be working properly. 

The next step is to mount everything inside the box enclosure. The placement of the plugs has been determined and we should be able to start routing the connection and cable paths in and out of the box soon. 

A small pack of buck converter boards has also been ordered along with a 24V power supply. This will allow us to shrink the amount of external plugs down to one. The raspberry pi will be powered internally from a 5V buck converter.

jeffrobots, 2016-03-25:
@cconklin 

I've just finished putting together the production controls circuitry and I've verified that it's working well. 

Last Monday we talked about what software changes we would need to make in order to move to a 3-relay hardware solution and you mentioned me not needing to change anything on my side (aside from the pin mapping, I suppose). 

Is this still the case? I was thinking of migrating all of the pin maps such that each control object is exposed to the same pin for the fan and the light respectively. They're already tied to the same pump, so no worries there. 

Currently, a control module is exposed to 3 seperate pins (the valve is currently still included and needs removed). If your plan is to use one control module, then we won't need to change any structure and I'll just need to remap the pins to the correct ones to play with the current hardware. If you still wanted the control modules to be seperate, then I'll need to make more extensive changes.

For my reference, we will want the pump on the A1 relay, the fans in parallel on the A2 relay, and the lights in parallel on the A3 relay. The power toggle line for the soil sensors will be on A0 under the sensor model.

cconklin, 2016-03-25:
@jeffrobots 

> Is this still the case?

Yes. I only have the web service interact with one control module (ID 1).

jeffrobots, 2016-04-12:
Work pertaining to this issue was completed at the end of last week on April 8th

closed 2016-04-12 by jeffrobots