ECESeniorDesign/greenhouse_envmgmt #24: Install and mount environment hardware

jeffrobots, 2016-03-30:
The following components need to be mounted or installed to the greenhouse box. 

- [ ] Greenhouse sensor PCBs
    * These need to be coated in acetate first.
    * Three PCBs should be populated with one as a backup.
    * Custom cabling will need to be created. This should be measured.

- [ ] Greenhouse controls box
    * See issue #23 for specifics.
    * Box should be mounted on the back or side of the greenhouse.

- [ ] Pump/Fans/Lights
    * The pump will need a water reservoir to draw from, preferably mounted beside or behind the greenhouse. 
        * The tank level sensor will need to run through the controls box and to the reservoir. 
        * A 1" diameter tube is needed to house the level sensor.
    * Fans need to be bolted to the sides of the greenhouse. 
    * Lights should be mounted on the top of the box and cabling should run so that the box can open

jeffrobots, 2016-03-30:
@cconklin 

My goal is to try very, very hard to have all of this done by Friday of next week so that we can begin integration and test. I'm not sure how close I'll be to that deadline, but I may need a bit of help mounting some of the hardware to the greenhouse.

As soon as everything is installed, I'll want to get the web service/application running and integrated by the end of the following week, which is April 15th, so that we have almost two full weeks to debug. The sooner the better, but this is the latest these tasks should be done in my mind. 

Let me know if you have anything blocking you currently in terms of getting the web service and application running on the Pi.

cconklin, 2016-03-30:
Currently nothing blocking me. I will try to get the web service running on the Pi by this Friday.

cconklin, 2016-04-01:
@jeffrobots 

Web service installed on the pi and runs correctly. I've looked into getting it to start automatically, and that shouldn't be any real issue.

cconklin, 2016-04-03:
I have the autostart for the pi working. I can get it set up on yours sometime this week.

jeffrobots, 2016-04-03:
Sounds good. 

Of course you'll get to see it during the status meeting tomorrow, but just to give you an update as to where I'm at in case you have any questions before the meeting.

* I've received the enclosure with all of the necessary holes routed into it. Standoffs were also made for the raspberry pi so that the RJ45 port isn't at the base of the enclosure, but rather in the middle.
* I've made a metric ton of custom cabling and I believe the internal enclosure cabling is complete.
    * This includes gluing the barrel jacks in place. 
* I've sorted out the general arrangement of all of the components and they're currently taped in place. 
    - [ ] I'd like to drill holes for standoffs for the remaining boards. 
        * One board has its standoff holes covered because I'm just that good. This one may need to be glued down. 
* I've calibrated the buck converters to 12V and 5V respectively. They appear to be pretty stable. 
    - [ ] I still need to actually power the raspberry pi via the cable I made to make sure it works. I've been putting it off.
* External cabling:
    * I've made a parallel cable to neatly hook up the fans. I may do this with the sensors as well.
    - [ ] I still need to hack together a cable for the LED lights so that they can use the barrel jack on the back of the box. 
    - [ ] I still actually need to make the cable to connect to the sensors. I will do this after we decide how to mount the sensors. 
* I've populated a second greenhouse sensor and verified that it works. I will do the third once everything else is together.

I'm in the lab right now for the next hour or so. I will finish putting together the cables that we need to connect to the controller box. If I have time after doing that, I'll tape everything into its final position in the box and try to fire it up using the 24V supply.

jeffrobots, 2016-04-03:
I've tested the raspberry pi in the enclosure running from the 5V buck converter. Everything works as intended. I also ran the controltest script and successfully toggled the correct relays. I'm not really worried about the sensors; I expect them to work with no issues. 

The last remaining hardware tasks include:

- [ ] Mounting the water level sensor inside a PVC tube and running cable to the box. 
    * We also need a tank. 

- [ ] Permanently mounting all of the hardware inside the box with standoffs/glue/etc. 

- [ ] Calibrating sensors

- [ ] Finishing up cabling. 
    * This is mostly done, but I'll probably need to change a few things once it's all mounted and I see where all of the components need to go.

cconklin, 2016-04-03:
The requirements specify a 5 gallon tank.

jeffrobots, 2016-04-12:
Issue is no longer relevant. Though some things still technically need to be mounted, this does not need an issue to keep organized any longer.

closed 2016-04-12 by jeffrobots